avr-rust/rust-legacy-fork #164: Could not compile core - LLVM Error: Cannot Select

jonahbron, 2020-04-08:
I just built the latest from the `avr-support-upstream` branch the other day (6a48e6e267bd295565633f4f432d1c809219df10).  When compiling another library (avr-hal), it tries to build `core` and fails.

```bash
$ XARGO_RUST_SRC=/run/media/jonah/data/Projects/avr-rust/src cargo +avr xbuild --target avr-unknown-unknown --verbose
WARNING: There is no root package to read the cargo-xbuild config from.
+ "rustc" "--print" "sysroot"
+ "rustc" "--print" "target-list"
+ "/home/jonah/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.s99JD96fQHru/Cargo.toml" "--target" "avr-unknown-unknown" "-v" "--" "-Z" "force-unstable-if-unmarked"
    Updating crates.io index
   Compiling compiler_builtins v0.1.26
   Compiling core v0.0.0 (/run/media/jonah/data/Projects/avr-rust/src/libcore)
     Running `rustc --crate-name build_script_build /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.26/build.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type bin --emit=dep-info,link -C opt-level=s -C codegen-units=1 --cfg 'feature="compiler-builtins"' --cfg 'feature="core"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=8f88709af7275f23 -C extra-filename=-8f88709af7275f23 --out-dir /tmp/xargo.s99JD96fQHru/target/release/build/compiler_builtins-8f88709af7275f23 -L dependency=/tmp/xargo.s99JD96fQHru/target/release/deps --cap-lints allow`
     Running `rustc --crate-name core --edition=2018 /run/media/jonah/data/Projects/avr-rust/src/libcore/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C codegen-units=1 -C metadata=0425a61751752881 -C extra-filename=-0425a61751752881 --out-dir /tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps --target avr-unknown-unknown -L dependency=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps -L dependency=/tmp/xargo.s99JD96fQHru/target/release/deps`
     Running `/tmp/xargo.s99JD96fQHru/target/release/build/compiler_builtins-8f88709af7275f23/build-script-build`
   Compiling rustc-std-workspace-core v1.99.0 (/run/media/jonah/data/Projects/avr-rust/src/tools/rustc-std-workspace-core)
     Running `rustc --crate-name rustc_std_workspace_core --edition=2018 /run/media/jonah/data/Projects/avr-rust/src/tools/rustc-std-workspace-core/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C codegen-units=1 -C metadata=1d5b7545d773d921 -C extra-filename=-1d5b7545d773d921 --out-dir /tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps --target avr-unknown-unknown -L dependency=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps -L dependency=/tmp/xargo.s99JD96fQHru/target/release/deps --extern core=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps/libcore-0425a61751752881.rmeta`
     Running `rustc --crate-name compiler_builtins /home/jonah/.cargo/registry/src/github.com-1ecc6299db9ec823/compiler_builtins-0.1.26/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C codegen-units=1 --cfg 'feature="compiler-builtins"' --cfg 'feature="core"' --cfg 'feature="default"' --cfg 'feature="mem"' -C metadata=a0e694beb56f771f -C extra-filename=-a0e694beb56f771f --out-dir /tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps --target avr-unknown-unknown -L dependency=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps -L dependency=/tmp/xargo.s99JD96fQHru/target/release/deps --extern core=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps/librustc_std_workspace_core-1d5b7545d773d921.rmeta --cap-lints allow --cfg 'feature="unstable"'`
   Compiling alloc v0.0.0 (/tmp/xargo.s99JD96fQHru)
     Running `rustc --crate-name alloc --edition=2018 /run/media/jonah/data/Projects/avr-rust/src/liballoc/lib.rs --error-format=json --json=diagnostic-rendered-ansi --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C codegen-units=1 -Z force-unstable-if-unmarked -C metadata=80d3f061118f5dfb -C extra-filename=-80d3f061118f5dfb --out-dir /tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps --target avr-unknown-unknown -L dependency=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps -L dependency=/tmp/xargo.s99JD96fQHru/target/release/deps --extern compiler_builtins=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps/libcompiler_builtins-a0e694beb56f771f.rmeta --extern core=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps/libcore-0425a61751752881.rmeta`
LLVM ERROR: Cannot select: 0x7fb3d80431b0: i16 = bitcast 0x7fb3d8076fa0
  0x7fb3d8076fa0: i8,ch,glue = CopyFromReg 0x7fb3d89fefe0:1, Register:i8 $r24, 0x7fb3d89fefe0:2
    0x7fb3d8751e80: i8 = Register $r24
    0x7fb3d89fefe0: i16,ch,glue = CopyFromReg 0x7fb3d83eb7c8:1, Register:i16 $r23r22, 0x7fb3d83eb7c8:2
      0x7fb3d80826c8: i16 = Register $r23r22
      0x7fb3d83eb7c8: i16,ch,glue = CopyFromReg 0x7fb3d8752498, Register:i16 $r21r20, 0x7fb3d8752498:1
        0x7fb3d83ebb08: i16 = Register $r21r20
        0x7fb3d8752498: ch,glue = callseq_end 0x7fb3d8b9c830, TargetConstant:i16<0>, TargetConstant:i16<0>, 0x7fb3d8b9c830:1
          0x7fb3d83eb488: i16 = TargetConstant<0>
          0x7fb3d83eb488: i16 = TargetConstant<0>
          0x7fb3d8b9c830: ch,glue = CALL 0x7fb3d83eb558, TargetGlobalAddress:i16<{ i8, i32 } (i32) addrspace(1)* @_ZN4core3num7flt2dec8strategy5grisu22max_pow10_no_more_than17h24eda0bf549557d4E> 0, Register:i16 $r23r22, Register:i16 $r25r24, RegisterMask:Untyped, 0x7fb3d83eb558:1
            0x7fb3d80376c0: i16 = TargetGlobalAddress<{ i8, i32 } (i32) addrspace(1)* @_ZN4core3num7flt2dec8strategy5grisu22max_pow10_no_more_than17h24eda0bf549557d4E> 0
            0x7fb3d80826c8: i16 = Register $r23r22
            0x7fb3d8076a58: i16 = Register $r25r24
            0x7fb3d807eed0: Untyped = RegisterMask
            0x7fb3d83eb558: ch,glue = CopyToReg 0x7fb3d807f140, Register:i16 $r25r24, 0x7fb3d84de068, 0x7fb3d807f140:1
              0x7fb3d8076a58: i16 = Register $r25r24
              0x7fb3d84de068: i16,ch,glue = CopyFromReg 0x7fb3d84de000:1, Register:i16 $r21r20, 0x7fb3d84de000:2
                0x7fb3d83ebb08: i16 = Register $r21r20
                0x7fb3d84de000: i16,ch,glue = CopyFromReg 0x7fb3d84ddf98, Register:i16 $r19r18, 0x7fb3d84ddf98:1
                  0x7fb3d84ddc58: i16 = Register $r19r18
                  0x7fb3d84ddf98: ch,glue = callseq_end 0x7fb3d84ddf30, TargetConstant:i16<0>, TargetConstant:i16<0>, 0x7fb3d84ddf30:1

              0x7fb3d807f140: ch,glue = CopyToReg 0x7fb3d8144f70, Register:i16 $r23r22, 0x7fb3d84de000
                0x7fb3d80826c8: i16 = Register $r23r22
                0x7fb3d84de000: i16,ch,glue = CopyFromReg 0x7fb3d84ddf98, Register:i16 $r19r18, 0x7fb3d84ddf98:1
                  0x7fb3d84ddc58: i16 = Register $r19r18
                  0x7fb3d84ddf98: ch,glue = callseq_end 0x7fb3d84ddf30, TargetConstant:i16<0>, TargetConstant:i16<0>, 0x7fb3d84ddf30:1

In function: _ZN4core3num7flt2dec8strategy5grisu19format_shortest_opt17hfcc742cc4fb3bf66E
error: could not compile `core`.

Caused by:
  process didn't exit successfully: `rustc --crate-name core --edition=2018 /run/media/jonah/data/Projects/avr-rust/src/libcore/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=s -C panic=abort -C codegen-units=1 -C metadata=0425a61751752881 -C extra-filename=-0425a61751752881 --out-dir /tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps --target avr-unknown-unknown -L dependency=/tmp/xargo.s99JD96fQHru/target/avr-unknown-unknown/release/deps -L dependency=/tmp/xargo.s99JD96fQHru/target/release/deps` (exit code: 101)
warning: build failed, waiting for other jobs to finish...
error: build failed
error: `"/home/jonah/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.s99JD96fQHru/Cargo.toml" "--target" "avr-unknown-unknown" "-v" "--" "-Z" "force-unstable-if-unmarked"` failed with exit code: Some(101)
note: run with `RUST_BACKTRACE=1` for a backtrace
```

When I built Rust, I simply edited the config.toml to include the experimental AVR target, and set the prefix.  Other than that, I ran `./x.py build && ./x.py install`; nothing special.

Is there another commit I should have compiled that can compile core, or something else I did wrong?

Logarithmus, 2020-04-21:
I've ran into the same issue, unfortunately. Trying to make my university project in Rust. The deadline approaches rapidly, so most likely I'll just pick C.

jonahbron, 2020-04-27:
@Logarithmus Sorry for the delay in seeing this, hope it's not too late.  If you use the older more stable `avr-support` branch, it should compile and work just fine.  My understanding is that the `master` branch is unstable right now because upstream is being merged down into this fork, and not everything has been fixed yet.

jonahbron, 2020-04-27:
I have a Gist I put together to go from zero to having a Blink program compiled.  Hope this helps.

https://gist.github.com/jonahbron/640a825dd9a526d3a1c24f2c20ca44ac

Logarithmus, 2020-04-28:
@jonahbron Thank you for help, but I've figured out it myself short after my previous comment.

jonahbron, 2020-05-17:
@Logarithmus glad you got it working!  It would be greatly helpful if you could post the solution you found here, in case anyone else encounters the same issue.

Rahix, 2020-05-17:
@Logarithmus, maybe the code I wrote for `avr-device` can help here?  I have a procedural macro to generate the IRQ definitions which essentially creates [this function definition](https://github.com/Rahix/avr-device/blob/169804d17a756d847904e712d189faf51dbfbdac/macros/src/lib.rs#L83-L94):

```rust
#[no_mangle]
pub unsafe extern "avr-interrupt" fn <vector_id>() {
    // ...
}
```

where `<vector_id>` is `__vector_1`, `__vector_2`, `__vector_3`, ...

The mapping of IRQ to ID can be found in the datasheet.

jonahbron, 2020-05-17:
FYI I gave this another shot today, commit ID 54014d0.  Still the same error.

```bash
$ XARGO_RUST_SRC="$(realpath ../avr-rust/src)" rustup run avr cargo xbuild --target avr-unknown-unknown --example leonardo-blink
WARNING: There is no root package to read the cargo-xbuild config from.
    Updating crates.io index
  Downloaded compiler_builtins v0.1.28
  Downloaded 1 crate (135.1 KB) in 0.74s
   Compiling core v0.0.0 (/run/media/jonah/data/Projects/avr-rust/src/libcore)
   Compiling compiler_builtins v0.1.28
   Compiling rustc-std-workspace-core v1.99.0 (/run/media/jonah/data/Projects/avr-rust/src/tools/rustc-std-workspace-core)
   Compiling alloc v0.0.0 (/tmp/xargo.cVO5liTsZqc7)
LLVM ERROR: Cannot select: 0x7f66c48f3bc0: i16 = bitcast 0x7f66c46d9e20
  0x7f66c46d9e20: i8,ch,glue = CopyFromReg 0x7f66c4a1b1d0:1, Register:i8 $r24, 0x7f66c4a1b1d0:2
    0x7f66c4073160: i8 = Register $r24
    0x7f66c4a1b1d0: i16,ch,glue = CopyFromReg 0x7f66c4041f08:1, Register:i16 $r23r22, 0x7f66c4041f08:2
      0x7f66c473d1c8: i16 = Register $r23r22
      0x7f66c4041f08: i16,ch,glue = CopyFromReg 0x7f66c4073778, Register:i16 $r21r20, 0x7f66c4073778:1
        0x7f66c4042248: i16 = Register $r21r20
        0x7f66c4073778: ch,glue = callseq_end 0x7f66c40d6890, TargetConstant:i16<0>, TargetConstant:i16<0>, 0x7f66c40d6890:1
          0x7f66c4041bc8: i16 = TargetConstant<0>
          0x7f66c4041bc8: i16 = TargetConstant<0>
          0x7f66c40d6890: ch,glue = CALL 0x7f66c4041c98, TargetGlobalAddress:i16<{ i8, i32 } (i32) addrspace(1)* @_ZN4core3num7flt2dec8strategy5grisu22max_pow10_no_more_than17h24eda0bf549557d4E> 0, Register:i16 $r23r22, Register:i16 $r25r24, RegisterMask:Untyped, 0x7f66c4041c98:1
            0x7f66c4da4890: i16 = TargetGlobalAddress<{ i8, i32 } (i32) addrspace(1)* @_ZN4core3num7flt2dec8strategy5grisu22max_pow10_no_more_than17h24eda0bf549557d4E> 0
            0x7f66c473d1c8: i16 = Register $r23r22
            0x7f66c46d98d8: i16 = Register $r25r24
            0x7f66c4951270: Untyped = RegisterMask
            0x7f66c4041c98: ch,glue = CopyToReg 0x7f66c49514e0, Register:i16 $r25r24, 0x7f66c4565a98, 0x7f66c49514e0:1
              0x7f66c46d98d8: i16 = Register $r25r24
              0x7f66c4565a98: i16,ch,glue = CopyFromReg 0x7f66c4565a30:1, Register:i16 $r21r20, 0x7f66c4565a30:2
                0x7f66c4042248: i16 = Register $r21r20
                0x7f66c4565a30: i16,ch,glue = CopyFromReg 0x7f66c45659c8, Register:i16 $r19r18, 0x7f66c45659c8:1
                  0x7f66c4565688: i16 = Register $r19r18
                  0x7f66c45659c8: ch,glue = callseq_end 0x7f66c4565960, TargetConstant:i16<0>, TargetConstant:i16<0>, 0x7f66c4565960:1

              0x7f66c49514e0: ch,glue = CopyToReg 0x7f66c40aa9f0, Register:i16 $r23r22, 0x7f66c4565a30
                0x7f66c473d1c8: i16 = Register $r23r22
                0x7f66c4565a30: i16,ch,glue = CopyFromReg 0x7f66c45659c8, Register:i16 $r19r18, 0x7f66c45659c8:1
                  0x7f66c4565688: i16 = Register $r19r18
                  0x7f66c45659c8: ch,glue = callseq_end 0x7f66c4565960, TargetConstant:i16<0>, TargetConstant:i16<0>, 0x7f66c4565960:1

In function: _ZN4core3num7flt2dec8strategy5grisu19format_shortest_opt17hfcc742cc4fb3bf66E
error: could not compile `core`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
error: `"/home/jonah/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/bin/cargo" "rustc" "-p" "alloc" "--release" "--manifest-path" "/tmp/xargo.cVO5liTsZqc7/Cargo.toml" "--target" "avr-unknown-unknown" "--" "-Z" "force-unstable-if-unmarked"` failed with exit code: Some(101)
note: run with `RUST_BACKTRACE=1` for a backtrace
```