avr-rust/rust-legacy-fork #166: LLVM "Can't select" error while compiling onewire crate

Gaelan, 2020-04-23:
While trying to compile the [onewire crate](https://crates.io/crates/onewire), I ran into this:

```
$ rustup run avr-toolchain xargo build --target avr-atmega328p --release
    Updating crates.io index
  Downloaded onewire v0.3.13
   Compiling byteorder v1.3.4
   Compiling onewire v0.3.13
LLVM ERROR: Cannot select: 0x7fd640068b38: i8 = ROL 0x7fd640067908
  0x7fd640067908: i8 = ROL 0x7fd640068f48
    0x7fd640068f48: i8 = ROL 0x7fd6400678a0
      0x7fd6400678a0: i8 = add nsw 0x7fd640067428, Constant:i8<-31>
        0x7fd640067428: i8,ch = load<(dereferenceable load 1 from %ir.0, !range !9)> 0x7fd63efb77a8, 0x7fd640067768, undef:i16
          0x7fd640067768: i16,ch = CopyFromReg 0x7fd63efb77a8, Register:i16 %0
            0x7fd640068ee0: i16 = Register %0
          0x7fd6400673c0: i16 = undef
        0x7fd640067560: i8 = Constant<-31>
In function: _ZN7onewire7ds18b2017MeasureResolution7time_ms17hd07d86a70e0fb2a8E
error: Could not compile `onewire`.

To learn more, run the command again with --verbose.
```

The symbol mentioned by the error is this `time_ms` method:
```rust
#[repr(u8)]
#[derive(Debug, Copy, Clone)]
pub enum MeasureResolution {
    TC8 = 0b0001_1111,
    TC4 = 0b0011_1111,
    TC2 = 0b0101_1111,
    TC = 0b0111_1111,
}

impl MeasureResolution {
    pub fn time_ms(&self) -> u16 {
        match self {
            &MeasureResolution::TC8 => 94,
            &MeasureResolution::TC4 => 188,
            &MeasureResolution::TC2 => 375,
            &MeasureResolution::TC => 750,
        }
    }
}
```

This presumably is an LLVM bug, but I'm filing here first because I have no idea how close the llvm fork here is to upstream, and I don't want to file a bug report for something that's already fixed there.

Gaelan, 2020-04-23:
Here's corresponding IR. It fails on avr-rust's llc, but compiles fine on LLVM master:

```llvm
@switch.table._ZN7onewire7ds18b2017MeasureResolution7time_ms17h58c3c3d7917b0300E = private unnamed_addr constant [4 x i16] [i16 94, i16 188, i16 375, i16 750], align 1

; onewire::ds18b20::MeasureResolution::time_ms
; Function Attrs: norecurse nounwind readonly
define i16 @_ZN7onewire7ds18b2017MeasureResolution7time_ms17h58c3c3d7917b0300E(i8* noalias nocapture readonly align 1 dereferenceable(1) %self) unnamed_addr addrspace(1) #3 {
start:
  %0 = load i8, i8* %self, align 1, !range !9
  %1 = add nsw i8 %0, -31
  %2 = lshr i8 %1, 5
  %3 = shl i8 %1, 3
  %4 = or i8 %2, %3
  %5 = sext i8 %4 to i16
  %switch.gep = getelementptr inbounds [4 x i16], [4 x i16]* @switch.table._ZN7onewire7ds18b2017MeasureResolution7time_ms17h58c3c3d7917b0300E, i16 0, i16 %5
  %switch.load = load i16, i16* %switch.gep, align 1
  ret i16 %switch.load
}

!9 = !{i8 31, i8 -128}
```

dylanmckay, 2020-05-17:
There's been a bunch of bugfixes in LLVM master, I expect this will be the cause.

I'm working on https://github.com/rust-lang/rust/pull/69478, which will update the LLVM fork. When that branch is ready, I expect this problem to be fixed.