avr-rust/rust-legacy-fork #167: Make fails at 10% following instructions on Fedora 32

sevagh, 2020-05-17:
I followed exactly the instructions here (https://github.com/avr-rust/rust#unixlinuxnix). I can't get past 20% in the `make` compilation.

This is one of the first errors that appears:

```
gmake[3]: Entering directory '/home/sevagh/repos/build/build/x86_64-unknown-linux-gnu/llvm/build'
[  0%] Building CXX object lib/MC/MCParser/CMakeFiles/LLVMMCParser.dir/AsmLexer.cpp.o
[  0%] Building CXX object utils/TableGen/CMakeFiles/obj.llvm-tblgen.dir/AsmMatcherEmitter.cpp.o
gmake[3]: Entering directory '/home/sevagh/repos/build/build/x86_64-unknown-linux-gnu/llvm/build'
[  0%] Building CXX object lib/MC/CMakeFiles/LLVMMC.dir/ConstantPools.cpp.o
gmake[3]: Leaving directory '/home/sevagh/repos/build/build/x86_64-unknown-linux-gnu/llvm/build'
gmake[3]: Entering directory '/home/sevagh/repos/build/build/x86_64-unknown-linux-gnu/llvm/build'
[  0%] Building CXX object lib/Support/CMakeFiles/LLVMSupport.dir/AArch64TargetParser.cpp.o
In file included from /home/sevagh/repos/rust/src/llvm-project/llvm/include/llvm/Demangle/MicrosoftDemangle.h:14,
                 from /home/sevagh/repos/rust/src/llvm-project/llvm/lib/Demangle/MicrosoftDemangle.cpp:17:
/home/sevagh/repos/rust/src/llvm-project/llvm/include/llvm/Demangle/MicrosoftDemangleNodes.h:14:17: error: found ‘:’ in nested-name-specifier, expected ‘::’
   14 | enum Qualifiers : uint8_t {
      |                 ^
      |                 ::
/home/sevagh/repos/rust/src/llvm-project/llvm/include/llvm/Demangle/MicrosoftDemangleNodes.h:14:6: error: ‘Qualifiers’ has not been declared
   14 | enum Qualifiers : uint8_t {
      |      ^~~~~~~~~~
```
There are many LLVM errors like this. I'm wondering if it's related to Fedora 32's packages being too new (since it was just released a few days ago).

I've attached a bigger output txt file: [output.txt](https://github.com/avr-rust/rust/files/4640692/output.txt)

Dependency versions:
```
sevagh:build $ g++ --version
g++ (GCC) 10.1.1 20200507 (Red Hat 10.1.1-1)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

sevagh:build $ cmake --version
cmake version 3.17.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
sevagh:build $ make --version
GNU Make 4.2.1
Built for x86_64-redhat-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

jonahbron, 2020-05-18:
I think I'm running into a similar issue, if not the same one.  Lots of errors from LLVM, but it stops at 22% for me.

https://pastebin.com/dq0XCFHr

Versions:

```bash
$ g++ --version
g++ (Clear Linux OS for Intel Architecture) 10.1.1 20200513 releases/gcc-10.1.0-36-gf2b77b928a
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cmake --version
cmake version 3.17.2

CMake suite maintained and supported by Kitware (kitware.com/cmake).
$ make --version
GNU Make 4.3
Built for x86_64-generic-linux-gnu
Copyright (C) 1988-2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

jonahbron, 2020-05-22:
@sevagh Not sure if you're still having issues with this, but I've gotten around it by creating a Docker container that can build Rust.  I'm really liking this approach because 1) I don't have to think about installing dependencies anymore, and 2) it will always stay the same unless I explicitly update it.

Two versions:

Generic Rust build env: https://hub.docker.com/r/jonahbron/rust-builder
Additional convenience for building AVR fork: https://hub.docker.com/r/jonahbron/avr-rust-builder

Post about it:
https://jonah.name/posts/building-rust/

sevagh, 2020-05-23:
@jonahbron works like a charm, thanks:
```
sevagh:repos $ rustup run avr rustc --version
rustc 1.37.0-dev
```

dylanmckay, 2020-06-09:
` enum Qualifiers : uint8_t {`

Perhaps the compiler is an older version that does not support specifying of `enum` size? This is a weird one. I imagine there will be an upstream LLVM bug to track this.

Here are my versions for reference:

```
cmake version 3.17.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).
1.10.0
cc (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

FWIW I'm on a rolling release distro, packages are generally always their most up-to-date version.

dylanmckay, 2020-06-09:
https://bugs.gentoo.org/show_bug.cgi?id=708730
https://github.com/facebook/hermes/issues/261